maholli/SAM32 #22: Have SAM32 connect to eduroam

maholli, 2019-05-16:
IoT projects can be difficult on university campuses when dealing with WPA2 enterprise access points.

maholli, 2019-05-16:
It is now possible to connect to eduroam and visitor wifi using the ESP32SPI library!

See @ladyada's example: [esp32spi_wpa2ent_simpletest.py](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/examples/esp32spi_wpa2ent_simpletest.py) 

You'll need to have your ESP32 flashed with the SAM32 NINA-FW firmware. [See the SAM32 ESP32SPI directory here](../tree/master/firmware/esp32_firmware/ESP32SPI)

ladyada, 2019-05-16:
is WiFiNINA 1.3.0 working for y'all?

maholli, 2019-05-16:
@ladyada yes I haven't ran into any issues. Something in particular not working on your end?

ladyada, 2019-05-16:
nope - just good to check :)

maholli, 2019-05-16:
oh! yes all good. 👍 

ESP32SPI is so helpful, I haven't needed anything else on the ESP32 co-processor in quite a while 😄